scanlime/fadecandy #46: Drive WS281x from Raspi directly (using rpi_ws281x)

dokterbob, 2015-03-12:
Quite recently @jgarff managed to implement control of NeoPixels directly from the Raspberry's GPIO. This should pave the way for having an OpenPixel control server that's running without use of Fadecandy.

https://github.com/jgarff/rpi_ws281x

What do you think about it? Could it be part of this project or should it be a distinct effort?

j4k3, 2015-03-12:
So... what does that have to do with the fadecandy?

dokterbob, 2015-03-12:
It might make it redundant (for some applications).

henkenli, 2015-03-13:
This might work if the need for speed is limited. The Raspberry is not an realtime system. 
Unless there is HW support for Times' GPIO sequences.

henkenli, 2015-03-13:
Should be "timed", not " Times'"

dokterbob, 2015-03-13:
That's precisely what rpi_ws281x is doing; they're using the two built-in PWM channels to perform DMA, similar to the Fadecandy firmware.

henkenli, 2015-03-13:
Interesting!
I suppose there is a limit to no of pixels etc.
How about the new gen of LED strips "Digital RGB LED" w 1,2 MHz clock from Adafruit?
Will that also work w RB?

Skickat från min iPhone

mpinner, 2015-08-23:
i was looking for just this. thanks!

RGB-123, 2015-08-23:
You may want to consider Yona's implementation here - https://github.com/Yona-Appletree/ledSPI that does just that.  Hardware hat here - https://twitter.com/RGB_123/status/614610227099385856