scanlime/fadecandy #51: fcserver crash on USB disconnect

ferrouswheel, 2015-08-11:
Working with my art project based on a raspberry pi, I disconnected USB cables from two fadecandies then plugged them back in. My neopixel strips wouldn't update after reconnecting.

Checking what happened via ssh, I was surprised to find that `fcserver` was no longer running. Checking the logs, there was no error message. The log registered the disconnection of the first fadecandy, but not the second so I assume it crashed somewhere in between.

I am running on a Raspberry Pi 1 Model B+, with a raspiado 4 port USB shield (and the fadecandiess plugged in via the shield).

```
$ uname -a
Linux 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l GNU/Linux
```

Restarting fcserver manually everything started working as expected.

I wanted to open this in case other people have experienced similar behaviour and we can get more evidence/context. I don't expect there is enough information to go on here, but I'll update this issue if I experience the same behaviour.

scanlime, 2015-08-11:
There's no guarantee the crash you're hitting is the same, but this seems like it could be an issue I've run into before. I tracked it down to a bug in libusb. There wasn't a way to work around that bug in fcserver as far as I could tell, and I didn't get around to filing a bug upstream in libusb :(

jlmeredith, 2015-11-08:
I have experienced this same issue on the art install I am working on currently.  

I understand the upstream restrictions on tracking a bug that has a limited reach.  Is there anyway we can add a parameter to fcserver to monitor for this condition and not exit but rather wait for a reconnection.  Or is this an issue of fcserver crashing due to the disconnect?  I have some programming experience but not anything at the level of your work Micah (I watched your videos on reverse engineering the DVD burner, I was captivated for a long while).

I am wondering if having fcserver run as a daemon and then use something like monit to restart the service if it fails?  I have had a similar issue in the past with a regular running service that processed a JSON feed.  It would regularly fail due to a segfault related to the version of Ubuntu that we were running.  We could not change the stack config due to a launch that was in progress, so we used monit to determine if the service was running and if not, we would restart the service.  It would check on a 1 min basis.

scanlime, 2015-11-08:
I usually use supervisord to run both fcserver and my own apps, it's the
most reliable way to restart if something goes wrong.