scanlime/fadecandy #52: Reversable OPC mapping in fcserver

ferrouswheel, 2015-08-21:
I was debugging why my pixel mapping was a little weird and had a "ahah" moment.

I was trying to set the end of the led strip as the first pixel. Unfortunately my pattern generation code is a bit dumb and expects the opc pixel layout files to have pixels monotonically ascending on the Z axis, in order to detect different parts of my art structure.

It'd be really neat to be able to specify a negative pixel count in the config to be able to reverse the order pixels turn up in the opc message.

I tracked down the pixel mapping to here:

https://github.com/scanlime/fadecandy/blob/master/server/src/fcdevice.cpp#L553

As an example, the following line in the "map" key:
```
[0, 0, 63, -64]
```

Would send data from opc channel 0, from pixels 0-64 in the opc message, to the first strip on a fadecandy board but reversed. So OPC pixel 63 would be the first pixel on the first strip of a fadecandy, OPC pixel 0 would be the 64th pixel on led strip.

Not sure if it'd be more intuitive way to do the mapping with a flag or something.

As always I'm doing this at last minute before a festival, but might be able to tackle this later in the week.

aintmichael, 2016-02-23:
Did anyone ever come up with a good solution to this? I have a 64x32 grid that, due to cabling limitations, is mounted upside-down. It would be great to be able to flip and reverse the whole thing in the json rather than in my sending app. Is this possible?

ferrouswheel, 2016-02-24:
@aintmichael Alas I never made progress on this.

ferrouswheel, 2018-01-15:
This looks to be implemented in 9181cec

closed 2018-01-15 by ferrouswheel