scanlime/fadecandy #53: reflashing the fadecandy

drio, 2015-08-27:
Hi, 

When I first connected my FC to the usb port of my RPI, the server was able to detect it but I was only able to light up three out of 60 pixels in my neopixel (only the first three ones were on) when using the testing links from the webapp (all on, 50% brightness, etc...). 

I then tried to load a new firmware ` $ sudo dfu-util -D fc-firmware-v107.dfu`. After that, the server wasn't able to detect the FC anymore. 

Is there anyway to reset the firmware to the original state?

Thanks!

scanlime, 2015-08-28:
Hi,

The problem you're describing doesn't sound like a problem with the
Fadecandy; it's likely a soldering defect or single LED defect in the
strip. You can cut off the first 4 LEDs or patch over the affected LED
(which could be either the last to light or the first that isn't lighting,
depending on the exact nature of the problem.)

If the DFU update failed, the Fadecandy should be starting up in bootloader
mode automatically. This would be indicated by a solid LED on the board,
and a different USB identifier. If this isn't happening for some reason,
you can force it by temporarily bridging RX and TX on the "hacker port"
with tweezers while applying power.

After this, you'll still need to use the dfu-util to install the firmware.
The latest in Github is the same the boards are manufactured with.

-m

bbulkow, 2017-04-03:
Hi, How long should the dfu-util process take? When I run on a raspberry pi, there is an immediate load of a couple of blocks, then it hangs. I have tried waiting 10 or 15 minutes with no progress, then restarted using the RX-TX trick ( which worked ).

The FC is not attached to anything other than USB at the moment.

Help appreciated.

TehAMP, 2018-05-08:
I have a FC board that had the LED fail to illuminate upon powering the board. I hear windows USB audio acks when plugging and unplugging it though. 

I shorted the TX and RX pads on the Hacker Port and got the LED to turn on, but I can't get the command `$ sudo dfu-util -D fc-firmware-v107.dfu.` to execute completely.

I was able to get the LED to stay on through booting the FC board by using ``$ sudo dfu-util -D fc-firmware-v107.hex.` but not the DFU file. 

But I still can't get my windows machine or RPI-W to see the FC board through localhost when running FC Server. Help!