scanlime/fadecandy #66: Add Support for RGBW LED

garrettdowd, 2016-03-09:
A few people are interested.

Discussion Page
[https://groups.google.com/forum/#!topic/fadecandy/HajFR96c-uc](url)

shootdaj, 2017-02-01:
How technical of a task would this be? Can a newbie (to FadeCandy) tackle this? I know C so I was thinking of taking a stab at it.

gerwitz, 2017-06-07:
+1 but my learning curve to contribute would be steeper than my readiness to contribute to a bounty.

jabdoa2, 2017-10-28:
This definitely works. Order or the channels is a bit weird by default (GRB) because of WS2812 order. You can either change than in the config or rearrange your OPC stream. We do the latter in the Mission Pinball Framework (https://github.com/missionpinball/mpf/) and it works fine. Guess this issue can be closed.

apathyuk, 2018-11-11:
+1 for having RGBW formally supported.  Be great if it autoselected the white LED over equal values of the RGB

jabdoa2, 2018-11-11:
@chromotechnotramp RGBW work with the fadecandy. Channel mappings will be off though which breaks features like white point correction. I guess this will be hard to fix with the current way the fadecandy works.