scanlime/fadecandy #68: FadeCandy not showing up or connecting with fcserver

reesej, 2016-04-25:
Hello,

I briefly saw fadecandy connected in fcserver when I first plugged my fadecandy in; I also heard the USB acknowledgement noise. When I checked localhost 7890 it did not show up. I then tried a second FadeCandy, and I did not hear the USB noise or see a connection in fcserver.

I then tried again on another machine, and the same thing happened again with both FadeCandys as did on the first machine.

Both computers are running Windows 10.

I have fadecandy package 2, and fadecandy MES '13, LED Controller fc64x8 Rev B with the miniUSB connection.

arminbw, 2016-07-14:
Hi,
I might suffer from the same problem. I have two fadecandies (firmware version 1.07) and I have tested them with two different usb cables on OSX and on a raspberryPi.

Sometimes the fadecandy gets detected after I connect it: the little led flashes, the fadecandy server finds it. **But most of the time the device is not detected.** It doesn't even show up in the OSX system profiler.

Any ideas?

reesej, 2016-07-19:
Hello,

For me my problem was first the USB cable. Apparently some USB cables are
charge only and no data. Weird. My next problem was that I was not using
the correct pins. I believe I was on pin 7 thinking it was pin 0. So  had
to either change the code to write to pin 7, or just move my led-data-in
back to the code default pin 0. Note that the fade candy will recognize the
LEDS but it will not wrote to them if you have the wrong pin. I am not
familiar with osx, but I got it to work on Windows 7 and Windows 10. I did
see some people had issues with the drivers for it though, on Windows and
osx. Good luck!

arminbw, 2016-07-21:
OK, glad you solved your problem! I guess I need to open a new issue now and you close yours.