scanlime/fadecandy #70: Bootloader support for MK20DX256 (Teensy 3.1/3.2)

DatanoiseTV, 2016-06-14:
Hi,

Did you work also on a bootloader for the Teensy 3.1 or 3.2?

Cheers

DavidMenting, 2016-08-09:
I wouldn't mind having a go at this one, but don't know where to start.

What would be needed to get this bootloader working on the MK20DX256? Is the only thing that needs changing the linker script?

scanlime, 2016-08-11:
AFAIR the changes are just memory map- the locked boot block has to be larger, since the locking granularity is a fraction of the total memory size rather than being a fixed size.

The actual changes to support the MK20DX256 are probably pretty quick; the main thing I'd been dragging my feet on was updating the build infrastructure to support binaries for multiple platforms. I didn't want to add that without committing to maintain it.

DavidMenting, 2016-09-22:
Would the things you describe simply be a matter of changing the memory section in the linker script like this?:
```
MEMORY
{
  BOOT_FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 8K
  APP_FLASH (rx) : ORIGIN = 0x00010000, LENGTH = 248K
  RAM  (rwx) : ORIGIN = 0x1FFF8000, LENGTH = 64K
  FLEXRAM (rwx) : ORIGIN = 0x14000000, LENGTH = 2K
}
```
Or is that way too naïve? I'll try flashing the resulting .bin file to a Teensy 3.2 tomorrow.

milkpirate, 2016-11-16:
Hey David,

I am pretty curious if your attempt to port to 3.2 worked out?

DavidMenting, 2016-11-16:
So far, I've merged a lot of code from Paul Stoffregen's recent Teensy code into Scanlime's bootloader. I've gotten it to compile without errors but I didn't get much further than that. 

I made a new linker file for the DX256 and added some DX256 specific sections to mk20dx128.c

The reflashed Teensy enumerates as a normal USB Teensy even if I force the check whether to enter the bootloader to always evaluate to true.

My fork with dx256-bootloader branch is over here: https://github.com/DavidMenting/fadecandy/tree/dx256-bootloader

I'm not sure how to proceed/debug from here onward. Any help is appreciated

ericsims, 2017-01-29:
I am working on this as well. I am able to flash my MK20DX256 on my own board. 

My problem is that I am getting a hardfault at Line 510 of usb_dev: `NVIC_ENABLE_IRQ(IRQ_USBOTG);`

I have double checked the IRQ table against the datasheet and cant find a problem.

Has anyone else experienced this with the dx256? I have been working on this for a few weeks now. I am really not sure where the issue is coming from.

Thanks.

DavidMenting, 2017-01-30:
Which hardware are you using? Can I try out your code somewhere?

ericsims, 2017-02-01:
Dave, the code I am using is your branch, where MK20DX256VLH7 the IRQ_USBOTG value is 73.

DavidMenting, 2017-02-02:
You might want to pull commit https://github.com/DavidMenting/fadecandy/commit/ea592e6aa63d5ff7850ac5a7d905e97999c9707e as I previously had errors in my linker script. I haven't flashed this version on my hardware yet, though.

ericsims, 2017-02-07:
neither ea592e6 nor f38e433 work for me :/. Im not sure why though. (btw, I think the libusb error is just an unrelated windows problem)

```
C:\Users\ericsims\fadecandy\bootloader>make install
openocd -f openocd.cfg -c "program fc-boot.hex verify reset"
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00498-gbbfb673 (2016-10-28-16:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : add flash_bank kinetis k40.flash
adapter speed: 1000 kHz
none separate
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Info : add flash_bank kinetis pflash.0
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 1000 kHz
Info : JTAG tap: k40.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Info : k40.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : JTAG tap: k40.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
Info : MDM: Chip is unsecured. Continuing.
k40.cpu: target state: halted
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Info : Probing flash info for bank 0
Info : Probing flash info for bank 1
Warn : Flash Configuration Field written.
Warn : Reset or power off the device to make settings effective.
wrote 6144 bytes from file fc-boot.hex in 0.173756s (34.531 KiB/s)
** Programming Finished **
** Verify Started **
Error: JTAG-DP STICKY ERROR
Error: Failed to write memory at 0xe0001040
Warn : target was in unknown state when halt was requested
k40.cpu: target state: halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0x1fff8216 msp: 0x20007fc0
Error: error executing cortex_m crc algorithm
Error: checksum mismatch - attempting binary compare
diff 0 address 0x00000400. Was 0xff instead of 0x07
diff 1 address 0x00000401. Was 0xff instead of 0x4b
diff 2 address 0x00000402. Was 0xff instead of 0x10
diff 3 address 0x00000403. Was 0xff instead of 0xb5
diff 4 address 0x00000404. Was 0xff instead of 0x4a
diff 5 address 0x00000405. Was 0xff instead of 0xf2
diff 6 address 0x00000406. Was 0xff instead of 0x02
diff 7 address 0x00000407. Was 0xff instead of 0x60
diff 8 address 0x00000408. Was 0xff instead of 0x4b
diff 9 address 0x00000409. Was 0xff instead of 0xf2
diff 10 address 0x0000040a. Was 0xff instead of 0x80
diff 11 address 0x0000040b. Was 0xff instead of 0x44
diff 12 address 0x0000040c. Was 0xfe instead of 0x0a
diff 13 address 0x0000040d. Was 0xff instead of 0x78
diff 14 address 0x0000040e. Was 0xff instead of 0x22
diff 15 address 0x0000040f. Was 0xff instead of 0xb1
No more differences found.
embedded:startup.tcl:454: Error: ** Verify Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 498
at file "embedded:startup.tcl", line 454
make: *** [install] Error 1
```

robstolarz, 2017-02-09:
Alright, having given things another try (and wiring VREGIN correctly) it booted properly and we actually got it to communicate with a computer. The only problem was that it didn't respond properly to the host, so it showed up in Windows as "Unknown USB Device (Device Descriptor Request Failed)" in Device Manager. Is there any reason this wouldn't work properly? e.g. since the IVT arrangement changed, we're not catching that interrupt or something?

scanlime, 2017-02-17:
It's possible the boot loader is seeing remnants of an app in flash.
Try forcing entry by holding tweezers between rx and tx while plugging in
USB.

DavidMenting, 2017-02-20:
I have had success getting the Kiibohd bootloader (https://github.com/kiibohd/controller/tree/master/Bootloader) working on my MK20DX256VLH7. The code isn't as clean as Scanlime's DFU bootloader, but it is working nonetheless.

robstolarz, 2017-02-23:
@scanlime We had already disabled the check for echo et al., opting to put a `true` in the conditional instead. The light came on and everything. But Windows just complains about the device. We don't know what's going wrong here.

dokterbob, 2017-05-19:
Any updates on this one? I have been buying 'old' Teensy 3.0's for LED projects for a while now (as I also want to be able to run other firmware than FC and like my extra pints) and would love to be able to upgrade to 3.2.

adammunich, 2018-10-08:
Has anyone made progress on this front? 

I managed to build @DavidMenting 's fork and am getting an odd behavior of my LED, it seems to "half-blink" (that is, blink to 50% brightness) with a frequency of about 2Hz. Does not seem right :-)

adammunich, 2018-10-08:
I managed to get mk20dx128.c working for the 256 -- the big errors were misconfigured clocks, and an interrupt vector table buffer that was too small.

now I have it enumerating as a fadecandy bootloader, but the serial is broken, and it's in a boot loop. 

still, progress :-)

adammunich, 2018-10-09:
hey folks i got it running, though I think the flash addresses are not correct...

https://github.com/adammunich/fadecandy_bootloader_mk20dx256

token47, 2020-11-10:
Any progress on this? Is there already a working FC in Teensy 3.1/3.2? Will start a project and only have a 3.2 at hand. Thanks!