scanlime/fadecandy #74: Specifying color order for device map

rogerguess, 2016-06-25:
Seems the order of my led's is GRB instead of RGB. Based on the example configs, I thought adding the string "grb" to the map would address this: 

```
{
    "listen": ["127.0.0.1", 7890],
    "verbose": true,

    "color": {
        "gamma": 2.5,
        "whitepoint": [1.0, 1.0, 1.0]
    },

    "devices": [
        {
            "type": "fadecandy",
            "map": [
                [ 0, 0, 0, 512, "grb" ]
            ]
        }
    ]
}
```

I get 'unsupported json mapping instruction'. What is the correct way to modify this?

MattGarfield, 2016-06-25:
The issue may be due to missing the "serial" JSON attribute which should come before "map" and if unique to each FC board. Try this (replacing the ### with your device serial number):

{
    "listen": ["127.0.0.1", 7890],
    "verbose": true,

    "color": {
        "gamma": 2.5,
        "whitepoint": [1.0, 1.0, 1.0]
    },

    "devices": [
        {
            "type": "fadecandy",
            "serial": "##################",
            "map": [
                [ 0, 0, 0, 512, "grb" ]
            ]
        }
    ]
}

rogerguess, 2016-06-25:
Good thought, but same issue unfortunately.

rogerguess, 2016-06-25:
Should the server software version be fcserver-1.02?

MattGarfield, 2016-06-25:
I believe so - fcserver-1.02 is the latest version

rogerguess, 2016-06-25:
OOOk... so I somehow downloaded something called fadecandy-package-02... I guess the adafruit tutorial is out of date or something. I don't see where I got that link:

https://codeload.github.com/scanlime/fadecandy/zip/master

Anyway... I downloaded the latest and how the server software version is now 1.04. The server config for color channel is working now!

closed 2016-06-25 by rogerguess

MattGarfield, 2016-06-25:
Good catch! Thanks for updating the thread. I'll be following your lead.