scanlime/fadecandy #78: P5js examples not running

shiqishen, 2016-07-29:
I'm trying to run the p5js examples with an 8x8 WS2812 neopixel matrix, but nothing is showing up.

shiqishen, 2016-07-29:
[image]

M-I-Kessler, 2016-07-29:
Did you run the example in a web browser?

shiqishen, 2016-07-29:
[image]

shiqishen, 2016-07-29:
yeah yeah nothing shows up for some reason

shiqishen, 2016-07-29:
The grid8x8_simple_noise.html runs in a web browser but when I run the javascript file in P5 nothing shows up, Why is this?

M-I-Kessler, 2016-07-29:
p5.js needs the sketch to be sketch.js or the html to be index.html. Also, p5 is very picky about how the file structure is setup.  I used p5.js to produce the sketches then transferred them out.  Can you open the browser console to see what the error is?  I'm not sure why it's not running from the browser.

M-I-Kessler, 2016-07-29:
Also make sure you have the libraries and images folders in the working folder.