scanlime/fadecandy #79: What does spacing in the ledGrid8x8(index, x, y, spacing, angle, zigzag) mean?

shiqishen, 2016-07-29:
In the library you didn't specify what spacing means. 
I'm trying to draw the 8x8 led grid on my image but for some reason it doesn't show up on the led.

shiqishen, 2016-07-29:
// Connect to the local instance of fcserver
var WebSocketAddress = "ws://127.0.0.1:7890"; 
//Show LED pixel locations.
var showPixelLocations = true; 
//Change the HTML Id of the canvas.
var canvasId = "sketch";

var weather;
var api = 'http://api.openweathermap.org/data/2.5/weather?q=new%20york';
var apiKey = '&APPID=1ccdf0562f24b2a4c4a4c178ba7fa1a5';
var units = '&units=metric';

var thunderstormimg;
var drizzleimg;
var rainimg;
var snowimg;
var atmosphereimg;
var clearimg;
var cloudsimg;

function setup() {
  createCanvas(400, 400);
  background(0);
  socketSetup(WebSocketAddress);
  function ledGrid8x8(index, x, y, spacing, angle, zigzag) {
 	ledGrid(0, 8, 8, width/2, height/2, spacing, 0, false);
  }
  
  thunderstormimg = loadImage("assets/thunderstorm.jpg");
  drizzleimg = loadImage("assets/drizzle.jpg");
  rainimg = loadImage("assets/rain.jpg");
  snowimg = loadImage("assets/snow.jpg");
  atmosphereimg = loadImage("assets/atmosphere.jpg");
  clearimg = loadImage("assets/clear.jpg");
  cloudsimg = loadImage("assets/clouds.jpg");

  setInterval(weatherAsk, 2000);
}

function weatherAsk() {
  var url = api + apiKey + units;
  loadJSON(url, gotData);
}

function gotData(data) {
  weather = data;
  var weatherr = weather.weather.main;
}

function draw() {
  
  if (weather) {
    var temp = weather.main.temp;
    var weatherr = weather.weather[0].main;
    fill(255, 90, 0);
    rect(0, 0, 20, temp*10);
    
    if(weatherr == "Thunderstorm"){
      image(thunderstormimg, 20, 10);
    }
    if(weatherr == "Drizzle"){
      image(drizzleimg, 20, 10);
    }
    if(weatherr == "Rain"){
      image(rainimg, 20, 10);
    }
    if(weatherr == "Snow"){
      image(snowimg, 20, 10);
    }
    if(weatherr == "Atmosphere"){
      image(atmosphereimg, 20, 10);
    }
    if(weatherr == "Clear"){
      image(clearimg, 20, 10);
    }
    if(weatherr == "Clouds"){
      image(cloudsimg, 20, 10);
    }
    drawFrame();
  }
}

shiqishen, 2016-07-29:
https://drive.google.com/folderview?id=0BzxLDzufuymqeFlrRk9WbkRIcW8&usp=sharing

M-I-Kessler, 2016-07-29:
Spacing is by pixels.  If your sketch says width/8 for the spacing and width was 200, then all you would be saying is the LED locations have 25 (500/8) pixels between them and where shown on screen.

shiqishen, 2016-07-29:
Thank you so much for answering all my questions!
I think I found why, in the browser console, it shows 
"Image from origin 'file://' has been blocked from loading by Cross-Origin Resource Sharing policy: Invalid response. Origin 'null' is therefore not allowed access."
I'm not sure what it means but I think there's something wrong with showing images, cause only the noise_simple and clouds sketch works. All your examples that involves pulling an image are not working.
Could you check on that and also take a look at my code? Thank you!!!

M-I-Kessler, 2016-07-29:
Use firefox and it should work.

M-I-Kessler, 2016-08-03:
Your files are not showing up on the google drive. only the opc.js file.  Also if you could get Gitter I could probably be more helpful.