scanlime/fadecandy #84: Simple Example - The function "registerDraw(OPC)" does not exist

toymachiner62, 2016-11-23:
I'm trying to follow your simple example using an LED strip (WS2812) and when I try to run the sketch in the processing IDE i get the error `"the function registerDraw(OPC)" does not exist`.

The error is on this line of OPC `parent.registerDraw(this);`

toymachiner62, 2016-11-23:
This helped point me in the write direction. Had to change a line in the OPC. https://groups.google.com/forum/#!topic/fadecandy/AMPRMHGqunE

closed 2016-11-23 by toymachiner62

bosgood, 2018-09-08:
For anyone else coming here with the same issue, the fix is to edit `OPC.pde:30` and replace:
```java
parent.registerDraw(this);
```
with:
```java
parent.registerMethod("draw", this);
```

toymachiner62, 2018-09-09:
@bosgood you should submit a PR for this fix!

bosgood, 2018-09-09:
@toymachiner62 I would! But it looks like it was already fixed a long time ago (6b80fdb136cc39d1da74d10f5e1e411f19533d8a), we must have gotten some old copies of this repo that were still floating around.