scanlime/fadecandy #96: Error Initializing USB library!

scdreger, 2017-08-22:
I've had fadecandy working on OS X no problem, but just tried to setup on a Raspberry Pi 3 running raspbian stretch. 

I followed all the directions in the Adafruit tutorial, but when I try to launch fcserver on the Pi, I get this line returned from terminal: "Error Initializing USB Library!"

Any help would be appreciated.

scanlime, 2017-08-22:
Is there any more to the error? Maybe permissions? Try running fcserver as
root and see if that helps, if so you can either continue running it as
root or you'll need to set permissions on the USB device.
-m

scdreger, 2017-08-23:
Thanks for the quick response ... That was the only error it reported. I did try running with sudo and the same error returned. 

I ended up doing a fresh flash and reinstalled FC and that worked. I've tested it out with some of my modified Processing examples I moved over from my mac. So far so good! Thanks again!