scanlime/fadecandy #108: libwebsocket init failed for relay

nurikabe, 2018-03-08:
Mentioned this in the forum, though pretty sure it's a bug so reporting here.

The new relay feature merged into master recently doesn't seem to be working on the Pi.

I have configured:

```
{
  "listen": ["127.0.0.1", 7890],
  "relay": ["192.168.1.142", 7891],
  "verbose": true
  ..
}
```
with another server listening from 192.168.1.142:7891.

Running fcserver with the above config, however, results in:

```
ERR: ERROR on binding to port 7891 (-1 99)
ERR: libwebsocket init failed for relay
```

plong0, 2018-03-12:
Same error for "listen" when I put an IP address other than null or the localhost (it works with 127.0.0.1 or its actual LAN IP address) on a Pi Zero W (Raspbian Stretch Lite)

```
ERR: ERROR on binding to port 7890 (-1 99)
ERR: libwebsocket init failed
```
with config
```
{
    "listen": ["192.168.15.3", 7890],
    "relay":  ["192.168.15.3", 7891],
    "verbose": true,
    ..
}
```

When I use null and connect a relay client, the fcserver on RPI crashes with a `Segmentation fault.`  Where running the same client examples connected to fcserver running on Mac OSX everything works as expected.

wmodes, 2018-08-18:
Same thing. Even with the default config on a Rpi. Config:
```
    "listen": ["127.0.0.1", 7890],
    "relay":  [null, 7891],
    "verbose": true,
```
Error:
```
$ ./fcserver-rpi config.json 
[1534618035:8677] ERR: ERROR on binding to port 7890 (-1 98)
[1534618035:8679] ERR: libwebsocket init failed
```
And with no config file:
```
$ ./fcserver-rpi
No system config file found, using default
[1534618109:8172] ERR: ERROR on binding to port 7890 (-1 98)
[1534618109:8173] ERR: libwebsocket init failed
```
Same results as root and as pi user.

abozzay, 2018-11-29:
> Mentioned this in the forum, though pretty sure it's a bug so reporting here.
>
> The new relay feature merged into master recently doesn't seem to be working on the Pi.
>
> I have configured:
>
> ```
> { [...]

I believe it is actually functioning as intended. "Relay" creates a second server on the host running fcserver, it does not create a client that connects to a server on another host. If you look at the error you are receiving, the (-1 99) indicates bind failed because "A nonexistent interface was requested or _the requested address was not local_." and based on your config and description, that is _not_ a local address. The other error mentioned in this thread (-1 98) indicates the port is already in use so another instance of fcserver is probably already running.

As for the other problems mentioned in this thread, the only valid values you should be able to pass to either "listen" or "relay" would be null, "127.0.0.1", or the ip of one of the interfaces on that host. That is because both create a server that waits for someone to connect to it, it is not creating a client that connects to another host.

Unfortunately that means the relay feature cannot be used to connect multiple instances of fcserver across hosts, which is what I was hoping it did.