scanlime/fadecandy #112: Weird flashes after a few reconfigures

jabdoa2, 2018-05-21:
We use the fadecandy as light platform in the Mission Pinball Framework to drive LEDs via OPC (see: http://docs.missionpinball.org/en/dev/hardware/fadecandy/index.html). After a few restarts of MPF we see weird flashing LEDs or wrong colors. Usually after around 20 restarts. MPF will reconfigure the fadecandy on restart (because it does not know that it already got configured before). Currently, the only way to fix this is to unplug the fadecandy which is very unpractical in a pinball machine. Any idea what could cause this and how to fix it?

SwizzleFish, 2018-05-21:
Getting the same issue on my project, after 20 or so restarts, Led's will get incorrect colors and get really flickery. Feels like its not getting fully reset. As Jabdoa mentioned, unplugging/plugging in the USB fixes the issue, but as this is a pinball machine, we need it to work over and over without need for this type of manual unplugging.

scanlime, 2018-05-21:
It may be related to an issue I've seen in the past but never got to the
bottom of. It seemed like it could have been related to memory corruption
in the color lookup tables, but I was able to rule that out. It's still
really a mystery to me, and my best guess is that some hardware glitch
(possibly noisy power) is causing memory corruption somewhere else, not
within the framebuffer or color lookup table, and then causing a persistent
problem with the color mapping.

In any case, the root cause *seems* to be electrical noise (on the 5V USB
rail or the shared ground), so that may be a starting point for debugging.

SwizzleFish, 2018-05-22:
Is there any test data I can provide to help? 

Its interesting that i've been able to run my project (with no changes, just playback) for long periods of time (multiple days) with no issues, but during development, the constant disconnect/reconnect seems to create what feels like "garbage-data" that just builds up.  In my case it starts to get worse at, lets say, 20 plays, with some colors being off and a light here or there being flickery.. but it continues to degrade, so at 30 plays, its an incoherent flickery mess..  Happy to provide video or anything to help!

scanlime, 2018-05-22:
The main thing that would help is a way to reproduce the problem reliably
and/or with a debugger attached.

-m

jabdoa2, 2018-05-22:
Reproducing it is no problem. I run two fadecandys and it usually happens at one of them first and the other one follows after a few more restarts. That probably also rules out the fadecandy server. I will try to create a script which sends exactly the same commands as MPF on restart.

jabdoa2, 2018-05-22:
For some reason I cannot easily reproduce it in a script. It sometimes happen when reconfiguring the fadecandy very often in a loop but it does not break reliably.

How can I attach a debugger to a fadecandy?

jabdoa2, 2018-05-23:
This reproduces the issue:

```
import opc, time, json, struct

numLEDs = 37
black = [ (0,0,0) ] * numLEDs
white = [ (255,255,255) ] * numLEDs
blue = [ (0,0,255) ] * numLEDs
red = [ (255,0,0) ] * numLEDs

def setGlobalColorCorrection(**obj):
    msg = json.dumps(obj)
    return struct.pack(">BBHHH", 0, 0xFF, len(msg) + 4, 0x0001, 0x0001) + bytes(msg, 'UTF-8')

for i in range(1000):

    client = opc.Client('localhost:7890')
    client.set_interpolation(True)
    client._socket.send(struct.pack(
            "!BBHHHB", 0x00, 0xFF, 0x0005, 0x0001, 0x0002, 0x00))

    client._socket.send(setGlobalColorCorrection(
        gamma = 2.5,
        whitepoint = [
            1.0,
            1.0,
            1.0,
        ],
        linearSlope = 1.0,
        linearCutoff = 0.0,
))

    for i in range(16):
      client.put_pixels(white, i)

    time.sleep(.001)
    for i in range(16):
      client.put_pixels(black, i)

for i in range(16):
      client.put_pixels(black, i)

numLEDs = 64
client = opc.Client('localhost:7890')

while True:
        for i in range(numLEDs):
                pixels = [ (0,0,0) ] * numLEDs
                pixels[i] = (255, 255, 255)
                client.put_pixels(pixels)
                time.sleep(0.01)
```
This will reproduce the issue for me after a few runs. Sometimes one run. Sometimes more. The effect is that other LEDs will also light up (and not only the chase). A bit random which ones but it happens.

jabdoa2, 2018-09-01:
Any progress on this? Anything we can help? This is really reproduceable for me.

cobra18t, 2020-05-16:
I believe I have the same problem as jabdoa2. I can reliably reproduce the problem by just restarting my computer. Until I unplug and replug the fadecandy USB, I get flickering during fade commands. Every once in a while, it will also replace a color (red becomes magenta or and orange-yellow becomes a greenish yellow). This occurs on both a Windows 7 laptop and a Lubuntu desktop.

AdrianD72, 2020-05-17:
I have had the problem too.  All the lights will dull in intensity and flicker.  It appears like it is a hardware issue but I disconnect the USB and and reconnect it and the problem goes away.

cobra18t, 2020-05-20:
To rule out an external hardware problem, I reduced my fadecandy installation down to the fadecandy, 40mm wires, and a strip of 3x WS2812B LEDs. I still get a flicker after a reconfigure. If I continue to reconfigure more times, the flicker gets worse and worse until it starts changing colors. Once I replug the USB, though, it is fine. In my intended application, replugging the USB is not a viable solution. Thanks for your help.
[image]

jabdoa2, 2020-05-20:
I looked into the firmware a bit more. It seems to me that this is a race between the ISR and the main loop. If the ISR fires while the fadecandy main loop is inside `fcBuffers::finalizeLUT()` and the ISR receives a LUT update (https://github.com/scanlime/fadecandy/blob/master/firmware/fc_usb.cpp#L93) it will store packets in `lutNew` AND free any packet which is already in there. It is then relatively likely that the next packet will received on USB reuses that memory area. When the ISR returns it will continue in `fcBuffers::finalizeLUT()` which might already have de-referenced a pointer into now freed/invalid memory (https://github.com/scanlime/fadecandy/blob/master/firmware/fc_usb.cpp#L138). The likelihood for the latter is not too high but keep in mind that we update three bytes per LED so there will be 255 packets send per update. @scanlime does that sound right?

carterhickman, 2020-09-09:
Did anyone figure out a solution to this issue? I am having this issue and can not solve it for the life of me.

jabdoa2, 2020-09-09:
Nobody fixed the code yet. Should not be too hard though. I failed to get the build system up for the firmware but I did not try too long either.

carterhickman, 2020-09-09:
I see. So do you know of any work arounds without changing the firmware?

jabdoa2, 2020-09-09:
Replugging the fadecandy after the race happened reliably fixes the issue. It rarely breaks at the first reconfigure. I guess adding a sleep before and after the reconfigure would also reduce the chance of this happening.

carterhickman, 2020-09-21:
I have found the cause of this issue in my situation. I was using opc.setColorCorrection() multiple times throughout my program to change brightness of the LEDs without changing the hue. Calling this function multiple times is what caused the bug for me.

jabdoa2, 2020-09-21:
Yeah this is one of the ways to trigger the memory corruption. Any changes to the color lookup table can cause this.

chrisgagne, 2021-03-12:
> I have found the cause of this issue in my situation. I was using opc.setColorCorrection() multiple times throughout my program to change brightness of the LEDs without changing the hue. Calling this function multiple times is what caused the bug for me.

I can confirm this is happening for me, too. Happened when using both Meanwell AC power supplies as well as Anker USB batteries for 5V power. 

Solving this would make it much easier for me to implement a maximum brightness algorithm that takes the color of each pixel into account.