scanlime/fadecandy #113: cmake for fadecandy/server doesn't work due to warning in libwebsockets/lib/base64-decode.c

krithin, 2018-05-25:
I followed the instructions in server/README.md to build the fadecandy server with cmake on my raspberry pi running archlinuxarm.

(I did this because I wanted to get a systemd unit file set up out of the box, so I could get the server to come up automatically whenever my fadecandy board was plugged into my pi's usb port).

I ran
$ make submodules
$ mkdir build
$ cd build
$ cmake ..
$ make

But then got this error:
[  3%] Building C object libwebsockets/CMakeFiles/websockets.dir/lib/base64-decode.c.o
/home/alarm/fadecandy/server/libwebsockets/lib/base64-decode.c:48:19: error: 'decode' defined but not used [-Werror=unused-const-variable=]
 static const char decode[] = "|$$$}rstuvwxyz{$$$$$$$>?@ABCDEFGHIJKLMNOPQRSTUVW"
                   ^~~~~~
cc1: all warnings being treated as errors
make[2]: *** [libwebsockets/CMakeFiles/websockets.dir/build.make:63: libwebsockets/CMakeFiles/websockets.dir/lib/base64-decode.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:190: libwebsockets/CMakeFiles/websockets.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

I believe the fix is a one-line change to the forked copy of base64-decode.c in scanlime/libwebsockets; i'm happy to send a PR to fix scanlime/libwebsockets, but would it be better to just have this depend on warmcat/libwebsockets's master branch instead of the scanlime fork?

krithin, 2018-05-25:
Sent https://github.com/scanlime/libwebsockets/pull/2

YannickdeHoop, 2019-06-03:
This also fails the build on Ubuntu Bionic x64