scanlime/fadecandy #120: Add support for FT232H (USB-> SPI) device to drive APA102/Dotstars

derekolson, 2018-11-01:
I've been working on a proof of concept to drive APA102 led strips with [FT232H](https://www.adafruit.com/product/2264) usb to spi boards. The high-speed interfaces are ideal for driving long runs of LEDs over usb. I've gotten this to work using libftdi and libmpsse, and I'm attempting to get this work within the fadecandy server ecosystem. WIP here - [ft232h-device](https://github.com/derekolson/fadecandy/blob/ft232h-device/server/src/ft232hdevice.h)

I'm having a bit of trouble getting the ft232h chip to work using only libusb commands and was wondering if there is anyone out there who can help or has interest in getting this working. Its eventually going to be a kind of a hybrid between the fcdevice and apa102spidevice. Thanks!

derekolson, 2020-08-24:
@embedded-creations Unfortunately, I kind of abandoned this fork - I was having trouble re-implementing the MPSSE commands that the FT232H needs in straight libusb code. (Admittedly, a novice in implementing any devices in libusb) I actually started building out another device agnostic LED framework in python using https://github.com/eblot/pyftdi - which turns out to be pretty performant in this application (unfortunately, haven't open sourced it yet). My latest endeavor is building out a full-on FPGA implementation that hooks into this framework. I may be able to help out where I can, if you want to try to take on this FT232H implementation.

embedded-creations, 2020-09-27:
Thanks for your response.  I'm going to try implementing something using the FT232H's FT245 mode, which should work with Fadecandy Server as it already has support for a FT245 DMX device.  I'll try to post back if I get it working, or at least I'll put a message on the Fadecandy mailing list