scanlime/fadecandy #123: Porting to Teensy 3.6

xudonax, 2018-12-20:
Would porting to the [Teensy 3.6](https://www.pjrc.com/store/teensy36.html) be doable? Or would this require a complete reimplementation? I'd be willing to take a swing at this if it would be somewhat doable.

With 16x as much RAM as the Teensy 3.0 (256K), more DMA channels and a lot more CPU power I'd say it would be a big improvement, at least in the number of supported LEDs and strips.

scanlime, 2018-12-20:
I haven’t been interested in porting Fadecandy because more ram or CPU
doesn’t necessarily help. The number of LEDs per strip has always been
limited mostly by the refresh rate necessary for fadecandy’s color
dithering.

-micah

xudonax, 2018-12-20:
Okay, understandable. But wouldn't the extra DMA channels enable more simultaneous strips per FadeCandy? I mean, I have a project planned that would require somewhere between 2500 and 4092 pixels per panel and I was hoping to get a bit less controllers than 7 or 8 FadeCandies it would need with the current hardware.

I'm still trying to get my head around how this actually works, so my apologies if this is a stupid question.

scanlime, 2018-12-20:
sure, it might make more channels per device possible- that’s just not even
the most compelling reason to do a hardware and firmware revision at this
point, and i’m not sure anything actually crosses the threshold for me
personally where it’s worth adding a firmware update process or additional
hardware models at this point. there are just so many other ways to drive
addressable LEDs that if Fadecandy isn’t right maybe something else is.

-m

xudonax, 2018-12-20:
Fair enough. Thanks!

closed 2018-12-20 by xudonax