scanlime/fadecandy #129: FadeCandy can't access raspberry pie

wugaoxue, 2019-04-21:
This is the first time I have used FadeCandy. I connected the raspberry pie with a USB cable, but the raspberry pie was unrecognizable. And the little lights on FadeCandy were not on. But when I connected FadeCandy to my computer, it worked very well. What's the matter with this? The voltage I use is 220V AC. I use it in China. I'm very grateful to anyone who can help me.

wugaoxue, 2019-04-21:
I just bought a Fade Candy. He can run well on my laptop. But it doesn't work on my raspberry pie. 
I inserted FadeCandy into my raspberry pie, and the raspberry pie couldn't recognize it.
I use this command to view the USB device I'm connected to. dmesg | tail -10
The following information appears

[ 42.691586] usb 1-1.1.2: new full-speed USB device number 7 using dwc_otg
[ 42.824940] usb 1-1.1.2: New USB device found, idVendor=1d50, idProduct=607a
[ 42.824955] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 42.824964] usb 1-1.1.2: Product: Fadecandy
[ 42.824973] usb 1-1.1.2: Manufacturer: scanlime
[ 42.824981] usb 1-1.1.2: SerialNumber: XXQKUQPZRFMGICJN
[ 43.671628] Under-voltage detected! (0x00050005)
[ 47.831631] Voltage normalised (0x00000000)
[ 143.511728] Under-voltage detected! (0x00050005)
[ 147.671925] Voltage normalised (0x00000000)

what should I do？I use hardware in China. Voltage in China is 220 V AC

EdwardDeaver, 2019-09-20:
Hi wugaoxue ,

What Operating System are you using on your Raspberry Pi?

What is the rated output of the power supply you are using for your Raspberry Pi?