scanlime/fadecandy #132: fcserver-osx won't run on macOS Catalina

sohanmurthy, 2019-10-12:
Running fcserver-osx on macOS Catalina (10.15) results in a `Bad CPU type in executable` error. This is likely due to Catalina dropping support for 32 bit applications. 😭

simonassank, 2019-11-29:
@scanlime is there any future plans for updating the software to x64?

mizraith, 2019-12-04:
Can confirm sohanmurthy's report on my 10.15 machine.   

Tried to make my own binary with no luck.     Please advise how to build or at least what to start banging on to try to get this working again.

sohanmurthy, 2019-12-10:
I'll preface all of this with the fact that I have no idea what I'm doing, but I was successfully able to compile and run `fcserver` on my 10.15 machine by changing the `-m32` values under [LDFLAGS](https://github.com/scanlime/fadecandy/blob/master/server/Makefile#L45) and [CPPFLAGS](https://github.com/scanlime/fadecandy/blob/master/server/Makefile#L46) to be `-m64`. 

I haven't tested lights yet, but it compiles, runs, and detects a board when connected via USB.

grahamma, 2019-12-15:
@sohanmurthy - i'll preface this all of this with the fact that I also don't really know what I'm doing, but i made the changes you describe to LDFLAGS and CPPFLAGS - unfortunately, when i run `make` to build the project, i get a whole bunch of `building for macOS-x86_64 but attempting to link with file built for macOS-i386` errors. Did you encounter that at all or did it just magically work for you? Are you using `make` to compile? Any other tips or insights would be appreciated. Thanks!

sohanmurthy, 2019-12-15:
@grahamma I didn't encounter the errors you described. To compile, I ran `make submodules` and then `make`.

Maybe first run `make clean` before attempting again?

platinumjesus, 2020-01-03:
@sohanmurthy solution worked for me too (running on Mac 10.15.2) - @scanlime do you think a good PR would be in the makefile to detect a build on Mac > 15 to apply different flags? 

Thank you btw, I am having so much fun with my Fadecandy Board.

aderrington, 2020-01-29:
Could someone please share a copy of their build for 10.15 please, I've been trying to do it but just keep having issues.

platinumjesus, 2020-02-02:
@aderrington here you go, I hope it works: https://drive.google.com/file/d/1xGxGz6mnrkcecOwndt47DhClBCKTre1V/view?usp=sharing

simonassank, 2020-02-02:
Just use the branch I've created and follow the guide how to build an executable https://github.com/scanlime/fadecandy/blob/master/server/README.md#getting-started.

elin05, 2020-12-03:
> @aderrington here you go, I hope it works: https://drive.google.com/file/d/1xGxGz6mnrkcecOwndt47DhClBCKTre1V/view?usp=sharing
That file no longer exists. Can you repost it please? Thanks!

einraum-design, 2020-12-11:
Same problem - thank you for the fix scanlime#132

platinumjesus, 2020-12-11:
@elin05 @einraum-design I am sorry as I no longer possess a Mac I am not able to build the package. It would be possible by adding the correct flags in the build command as mentioned by @sohanmurthy

einraum-design, 2020-12-11:
The fix with the correct flags works - thank you.

dplakon, 2020-12-14:
Here's another link:
https://drive.google.com/file/d/1wWPHWP1jnRp7uYM208nsG14CEjVhlRFu/view?usp=sharing

Only for OSX, binary file is /server/fcserver

orgicus, 2021-03-30:
If it helps, I've also uploaded a binary [here](https://sensori.al/fadecandy/fcserver-osx).
Note that once downloaded you may need to change permissions to make it executable for your user.
To do so :

1. Open Terminal
2. Type `chmod +x ` (space after chmod, space after +x)
3. Drag and drop the downloaded `fcserver-osx` file and Hit Enter.

"Should" (famous last words) be able to run it after doing this (if it didn't work the first time).

To compile from source you will need a c++ compiler and make installed as a minimum. 
Generally, if you have XCode Build Tools you should be good to go using Terminal: 

1. clone the repo along with dependencies (recursively): `git clone https://github.com/scanlime/fadecandy --recursive` (this will pull libusbx, libwebsockets and rapidjson)
2. navigate to the server folder: `cd fadecandy/server` (enter repo and server folder within it)
3. build submodules (dependencies): `make submodules`
4. build fadecandy server: `make` (this should output the `fcserver` executable in the server folder: feel free to move it as required)

The steps are pretty much from [server readme](https://github.com/scanlime/fadecandy/tree/master/server#getting-started)

An alternative for more advanced users is CMake.:

0. install CMake (if it isn't already installed): (this can be easily via [Homebrew](https://brew.sh/) or [CMake-GUI](https://cmake.org/download/) (as long the `cmake` command has been added to the `PATH` environment variable during install)
1. download the repo recursively and navigate to the `server` as listed above
2. make a build folder and enter it: `mkdir build && cd build`
3. run `cmake` to generate Makefiles: `cmake ..` (this will use the CMakeLists.txt file from the server folder above (hence ..))
4. run `make` to build: this will output `fcserver` in the current build folder

The same thing can be done using CMake-GUI :
1. press `Browse Source...` and select the path `fadecandy/server` folder
2. press `Browse Build...` and select the path to `fadecandy/server/build` (if the build folder doesn't exist, but the rest of the path is correct CMake-GUI will ask if you want it to make this folder for you)
3. press Configure: If you have XCode installed and want to compile this way feel free to try it (I haven't tried that option though), otherwise with XCode build tools, Unix Makefiles should work
4. press Generate to generate the xcode project or Makefiles 
5. compile (either using XCode or navigating to the `fadecandy/server/build` golder in Terminal and running `Make`)

Hope I didn't miss anything and hope this helps :)

ultra-vi0let, 2021-07-20:
@orgicus @dplakon @sohanmurthy @platinumjesus one thousand thanks to all of you.

so often its the dumb little things you overlook. I tried every which way to get this to work only to find out that in the end I was still trying to run the old build of the server because the newly built one was actually ending up inside of fadecandy/server/ instead of fadecandy/bin/, which of course was mentioned but I totally missed it. Sheesh lol.

Downloading the precompiled executable from the links in this thread didn't work for me. So it's likely you'll need the build tools that @orgicus mentioned so you can compile it yourself.

I suspect that some may have failed to change the makefile before attempting this. Many in this community aren't super fluent in the technical side (like myself) so I'll be slightly verbose here.

To anyone who has made it this far and still having trouble, this is what worked for me using `make`.

1. Open a terminal shell and get a fresh clone of the repo along with dependencies (recursively): `git clone https://github.com/scanlime/fadecandy --recursive` It will download to your home folder.
2. Use whatever method you want to navigate to `~/fadecandy/server` and edit the file `MakeFile` as detailed in [this reply](https://github.com/scanlime/fadecandy/issues/132#issuecomment-564298422) above : change `-m32` to `-m64` in lines `45` and `46`. Save the file.
3. If it's not already, point your terminal shell at the server folder: `cd fadecandy/server`
4. Now make the submodules with your updated MakeFile: `make submodules`
5. Finally, you can now build your new fcserver: `make`
6. The new executable will end up in the same folder you're running the command from, so you may want to move it to `fadecandy/bin` where the other executables are and rename it as `fcserver64` or something similar.
7. You should now be able to run a fadecandy server on a Mac running Catalina by executing that file. 

Another trivial tip I'll include is to just make sure you're working with the right folders and files if you are disorganized like I am and have multiple instances of the fadecandy repo living on your computer.

Once again thank you very much to everyone in this thread. I really needed to get this going again!